area515/Photonic3D #309: shutdown and other questions

jschoch, 2017-03-13:
first, is this the best place to ask questions or should I use the photocentric forum?

I'm building my own printer and wanted to know a few things.

how do I ssh to the pi?

How do i get more options then the "end user" configuraiton?

Assuming I can't ssh and there is no shutdown command in the GUI, how do I do a clean shutdown and avoid any FSCK issues?

WesGilster, 2017-03-14:
The forum is a good place to ask questions related to printing, configuration and other general questions. If you are quite sure you have a bug that needs to be researched, you could post it here.

You can still ssh to the pi as you would any other Linux box. If you changed the password during the install process, you would use that password instead of the typical "raspberry" password. "pi" is still the username. Other distributions of Photonic3D do have other passwords.

I'm not sure what end-user options you are looking for, or how deep you are interested in going. I suppose this might be a good set of guidelines:

Who I am:
I'd like to stay in the web GUI where I'm comfortable.
What I can do:
There is an endless supply of end-user options in the GUI. From gcode templates, printable customizers, advanced customizers, 2d -> 3d options, printer controls, calibration settings, ink configurations, projector autodetection, wifi setup options, Javascript calculators. It just goes on and on...

Who I am:
I have no idea or interest in any form of programming, but I don't mind editing configuration files. 
What I can do:
A great place to start is in /opt/cwh/config.properties. This gives you the ability to control what things are available inside of the GUI itself and what high level functions are available. Another end user editable file is: /opt/cwh/log4j2.properties which Photonic3d uses to give you the ability to present logs in a number of ways: https://logging.apache.org/log4j/2.0/manual/configuration.html#Properties

Who I am:
I'm a power user that likes to dabble in HTML and JS.
What I can do:
Browse around in the /opt/cwh/resourcesnew directory and you'll find all of the front end files for Photonic3d. In this area you'll find tons of new ways to present options in Photonic3d to the end-user.

Use the following soft commands for a "clean" shutdown.
sudo /opt/cwh/stop.sh
sudo halt

jschoch, 2017-03-14:
odd, i can't connect to port 22.  i'm running nmap with these results

```
Starting Nmap 7.01 ( https://nmap.org ) at 2017-03-14 00:32 PDT
Nmap scan report for photonic3d.local (192.168.1.86)
Host is up (0.0052s latency).
Not shown: 997 closed ports
PORT     STATE SERVICE
53/tcp   open  domain
5001/tcp open  commplex-link
9091/tcp open  xmltec-xmlmail
MAC Address: B8:27:EB:CC:C9:92 (Raspberry Pi Foundation)
Device type: general purpose
Running: Linux 3.X|4.X
OS CPE: cpe:/o:linux:linux_kernel:3 cpe:/o:linux:linux_kernel:4
OS details: Linux 3.2 - 4.0
Network Distance: 1 hop

```

WesGilster, 2017-03-14:
Yeah, that's odd, I thought that all of our installations keep sshd running. How did you install Photonic?

jschoch, 2017-03-15:
from the image on the github page from my recollection.   I can close if you don't think this is a bug.

jmkao, 2017-03-15:
That's odd, those images should have sshd installed, insofar as I don't specifically remove it from the jessie-lite base image in the build script.

WesGilster, 2017-06-28:
This issue has been opened for quite some time. Is it reproducible? If no response, I'm going to close.

WesGilster, 2018-01-21:
Photonic doesn't shut off SSH, so clean shutdown is available.

closed 2018-01-21 by WesGilster