area515/Photonic3D #311: grbl status "?" fails and other issues.

jschoch, 2017-03-17:
Trying to get grbl status "?" causes error.

My guess would be the params are not escaped which would create some possible security issues?

Another problem is the gcode responses do not get sent back to the client.  Is there active work on trying to fix this?

Finally, the gcode box does not submit on <enter>, perhaps it needs to be in a form tag or a js event needs setup?

Also Ctrl-x does not work from the g-code box which seems odd.

WesGilster, 2017-03-17:
> Trying to get grbl status "?" causes error.

Fixed this in my dev repo.

> Another problem is the gcode responses do not get sent back to the client. Is there active work on trying to fix this?

I've never seen this not work. You are saying the 'ok's aren't appearing?

> Finally, the gcode box does not submit on , perhaps it needs to be in a form tag or a js event needs setup?

Fixed this in my dev repo.

> Also Ctrl-x does not work from the g-code box which seems odd.

Seems to work fine for me. What browser are you using?

jschoch, 2017-03-22:
I'm using chrome, hope to test the fixes today.

jschoch, 2017-03-24:
? seems to work, and cut/paste also seems to work now.  I still get delays in the output from the gcode msgs.  The Ok's appear but sometimes I need to run "$$" several times to see the output.  A command history would be really nice, should i open a feature request?  I'd also like to see some way to stop the printer from the printer controls page.  Right now you need to "return to printer" and then re-select the printer and finally click the stop printer button.  

another possible bug is that the shutter stays open when you cancel a job, seems canceling the job should set shutter false.  

Let me know if you are open to the other things and i'll open separate issues for them.

WesGilster, 2017-03-30:
I'd need you to upload a diagnostic for me to know an example of something that isn't working. It isn't clear from your post. Command history should already exist for any button press in the gui...

Sure, open separate feature requests for the issues you'd like to see...

WesGilster, 2017-05-20:
Closing since there hasn't been activity on this and other features opened for other functionality.

closed 2017-05-20 by WesGilster