area515/Photonic3D #312: Dell 2300MP beamer not supported by Photonic3D?

Momboz52, 2017-03-29:
Hi
I am trying to understand and to use the Photonic3D concept and one of the issues is that my beamer is not recognized.
Does this correspond to the reality or am I wrong.

Thanks for confirming (or not confirming).

WesGilster, 2017-03-30:
Photonic3D will support any projector(like any print host software).

However, if you want to be able to control it's on/off features, bulb hours and autodetection capabilities through a serial port, you need to add them yourself or work with us to get those features available. 

Then, (unlike other print host software) it would be autodetected from that point forward.

Momboz52, 2017-03-30:
In my environment, it seems that Photonic3D makes some checking with the projector and then decided that there is no projector connected (although there is one connected, of course). When configuring a new printer, I have seen that Photonic3D sends some control characters on the projector serial interface and these are in hex:
7E 50 4E 0D 7E 71 56 0D 2A 20 30 20 4C 61 6D 70 20 3F 0D 07 14 00 05 00 34 00 00 11 00 5E
And then says: “Couldn't determine model of projector on port:COM5”.  I know that my projector sends characters on this serial interface as well.
I have difficulties to understand the situation. Did I get a wrong configuration?
By the way, I would be glad to contribute and work with you on this project.

WesGilster, 2017-03-31:
Photonic3D doesn't say that a projector isn't connected. Any application can determine there is something connected, but Photonic3d will attempt to determine the model of the projector that is connected. If your model isn't supported, then you need to add it like the others::
[
  {
    "name": "Optoma (HD141x)",
    "defaultComPortSettings": {
      "speed": 9600,
      "databits": 8,
      "parity": "None",
      "stopbits": "One"
    },
    "onHex": "7E303030302031",
    "offHex": "7E303030302032",
    "bulbHoursHex": "7E30303135302031",
    "bulbHoursResponseRegex": "(?s:OK(?:0|1)(\\\\d{4}.*))",
    "bulbHoursConversion": "ASCII"
  },
  {
    "name": "Vivitek (D535)",
    "defaultComPortSettings": {
      "speed": 9600,
      "databits": 8,
      "parity": "None",
      "stopbits": "One"
    },
    "onHex": "7E504E0D",
    "offHex": "7E50460D",
    "bulbHoursHex": "7E714C0D(function (){java.lang.Thread.sleep(100);})()",
    "bulbHoursResponseRegex": "[^\\\\d]*(\\\\d+)\\\\x0D?",
    "bulbHoursConversion": "ASCII",
    "detectionHex": "7E504E0D(function (){java.lang.Thread.sleep(1);})()7E71560D",
    "detectionResponseRegex": "(?s:.*DP3406.*\\\\x0D?)"
  },
  {
    "name": "Acer",
    "defaultComPortSettings": {
      "speed": 9600,
      "databits": 8,
      "parity": "None",
      "stopbits": "One"
    },
    "onHex": "4F4B4F4B4F4B4F4B4F4B0D",
    "offHex": "2A2030204952203030320D",
    "detectionHex": "2A2030204C616D70203F0D",
    "detectionResponseRegex": "(?s:.*?)(?i:Lamp) [10](?s:.*?)"
  },
  {
    "name": "ViewSonic (pjd7820hd)",
    "defaultComPortSettings": {
      "speed": 115200,
      "databits": 8,
      "parity": "None",
      "stopbits": "One"
    },
    "onHex": "0614000400341100005D",
    "offHex": "0614000400341101005E",
    "bulbHoursHex": "0714000500340000150163",
    "bulbHoursResponseRegex": "\\\\x05\\\\x14\\\\x00\\\\x06\\\\x00\\\\x00\\\\x00([\\\\x00-\\\\xFF]{2})\\\\x00\\\\x00[\\\\x00-\\\\xFF]",
    "bulbHoursConversion": "LittleEndian",
    "detectionHex": "071400050034000011005E",
    "detectionResponseRegex": "(?s:.*(?:\\\\x05\\\\x14\\\\x00\\\\x03\\\\x00\\\\x00\\\\x00[\\\\x00\\\\x01][\\\\x17\\\\x18]|\\\\x03\\\\x14\\\\x00\\\\x00\\\\x00\\\\x14)\\\\x0D?)"
  }
]

WesGilster, 2017-06-28:
Closed due to inactivity

closed 2017-06-28 by WesGilster