area515/Photonic3D #313: Just motors control or just projector control over Photonic3D

Momboz52, 2017-03-31:
It is possible to have an "ORed" configuration for a Photonic3D printer?
I mean, it is possible to have just the motors serial port active without any projector serial port, or
vice versa? 
The reason I am asking this is that I am not interested in the projector control but just the motors control.
Here is the page I am talking about:

[image]

WesGilster, 2017-03-31:
> printercfg

If that's what you want, then don't setup a projector at all. Just 
plugin your hdmi cable. Then setup your display to last available display.

Thanks,

Wes G.

Momboz52, 2017-03-31:
Hi Wes
If my understanding is correct, you are proposing to configure the printer this way:

[image]

However if I do so, and then try to start the printer it lands in a never ending loop and doesn't start.

Did I understand you correctly?

Many thanks

WesGilster, 2017-04-01:
Yep, that looks about right. I'm not sure what you mean by landing in a loop. Download a diagnostic after it fails to start and attach to this bug.

Momboz52, 2017-04-01:
Dear Wes
I thought a video could more than many words.
Then here a video that shows you both situations.
1st one: I created a simulated display with no port selected and in this configuration I can start and stop the printer without any problem. This is 1st part in the video.
2nd one: I created a simulated printer with selecting a motors serial port and in this configuration I can't start the printer. The only way to get out of this situation, I have to stop the host and restart it again. This is second part in the video.
You have been telling me about a diagnostic log, where can I find it?

[Video.zip](https://github.com/area515/Photonic3D/files/887817/Video.zip)

WesGilster, 2017-04-01:
The 'download' diagnostic button is on almost every screen in Photonic3d including the screen shot that you already uploaded. Just click it, and upload the file after you've attempted to start the printer.

Momboz52, 2017-04-01:
Here is the desired information

[image]
[image]
[image]
[image]
[LogBundle.zip](https://github.com/area515/Photonic3D/files/887891/LogBundle.zip)

WesGilster, 2017-04-02:
Most likely the com port settings for your printer are wrong because your printer won't respond to gcode. It actually is starting your printer after the two minute timeout though. If you know your settings are correct, and want to decrease that time, you can change it in the:

> Advanced Properties: Printer response timeout (ms)

Here we send the gcode:
```
2017-04-01 20:55:04,170 INFO o.a.r.s.SerialManager [qtp2222479-17] Attempting to assign firmware using serialPort:COM1 with settings:{Port:COM1 speed:115200 databits:8 parity:None stopbits:One handshake:None} to printer:CWH Template Printer(printerFirmwareSerialPort:null, projectorSerialPort:null Display:Simulated display:0)
2017-04-01 20:55:04,170 INFO o.a.r.s.SerialManager [qtp2222479-17] Attempting to autodetect resources for:PrinterFirmware using serialPort:COM1 with settings:{Port:COM1 speed:115200 databits:8 parity:None stopbits:One handshake:None} to printer:CWH Template Printer(printerFirmwareSerialPort:null, projectorSerialPort:null Display:Simulated display:0)
2017-04-01 20:55:04,170 INFO o.a.r.s.SerialManager [qtp2222479-17] Resources detected:{comPort:COM1 model:null settings:{Port:COM1 speed:115200 databits:8 parity:None stopbits:One handshake:None}}
2017-04-01 20:55:04,342 INFO o.a.r.p.Printer [qtp2222479-17] Firmware serial port set to:COM1
2017-04-01 20:55:04,373 DEBUG o.a.r.d.PrinterDisplayFrame [AWT-EventQueue-0] Blank realized:203
2017-04-01 20:55:05,435 INFO o.a.r.g.GCodeControl [qtp2222479-17] Write: G91
```

Here you tried starting another printer while the first printer was trying to start...
```
2017-04-01 20:56:45,626 DEBUG o.a.r.p.PrinterManager [qtp2222479-12] Attempting to start printer:CWH Template Printer
2017-04-01 20:56:45,626 ERROR o.a.r.s.PrinterService [qtp2222479-12] Error starting printer:CWH Template Printer
org.area515.resinprinter.job.JobManagerException: This printer:CWH Template Printer is being started. Can't start again.
```

Here is where it finally timed out after two minutes
```
2017-04-01 20:57:05,461 INFO o.a.r.g.GCodeControl [qtp2222479-17] lineRead: null
2017-04-01 20:57:05,461 INFO o.a.r.p.Printer [qtp2222479-17] Firmware Welcome chitchat:
2017-04-01 20:57:05,461 INFO o.a.r.s.SerialManager [qtp2222479-17] Completed assignment of firmware using serialPort:COM1 with settings:{Port:COM1 speed:115200 databits:8 parity:None stopbits:One handshake:None} to printer:CWH Template Printer(printerFirmwareSerialPort:COM1, projectorSerialPort:null Display:Simulated display:0)
2017-04-01 20:57:05,461 DEBUG o.a.r.p.PrinterManager [qtp2222479-17] Assigned 3dprinter firmware:COM1 to:CWH Template Printer(printerFirmwareSerialPort:COM1, projectorSerialPort:null Display:Simulated display:0)
2017-04-01 20:57:05,461 INFO o.a.r.p.PrinterManager [qtp2222479-17] Printer started:CWH Template Printer(printerFirmwareSerialPort:COM1, projectorSerialPort:null Display:Simulated display:0)
```

WesGilster, 2017-06-28:
This timeout is fairly adequately explained in the above logs. Feel free to reopen if you disagree.

closed 2017-06-28 by WesGilster