area515/Photonic3D #314: Printer and projector availability checking

Momboz52, 2017-04-05:
Hi all

Is there any possibility to suppress the availability checking for printer/projector when creating a printer configuration for Photonic3D?
I mean an option where Photonic3D doesn't care whether a printer/projector is connected or not. Just send the information to those and exactly if there were there and connected.

Many thanks / Momboz52

WesGilster, 2017-04-06:
Sure, that's what the simulated gcode responder, and the simulated display are for. You can use those and run your prints as if everything was connected.

Momboz52, 2017-04-06:
Hi Wes.

Yes, You are wright. This is what I have been doing so far.

However, my question is somehow different. I would like to do some testing and I would like that Photonic3D accept my serial port setting for motor/projector, but do not check whether this is true or not.
I wish that Photonic3D believes me when I tell it that there a printer/projector connected to serial port COMx/COMy and send commands normally to those.

I am wondering whether I have been clearer in asking my question. But we will see.

Many thanks / Mohamed

WesGilster, 2017-04-07:
What you are asking is clear, but doesn't make much sense. However, Photonic3d is still doing what you are asking.

You are asking that Photonic3d ignore the fact that your printer isn't setup on COMx even though it's set that way. What I'm telling you (in #313) is that you need to decide how long Photonic 3d should wait to determine your printer is missing and then ignore the fact there isn't a printer available. Currently that time period is set to wait 2 minutes for a response.

The way I understand it, is that you are saying 1 of the following:
1. 2 minutes is way too long to wait for your non-existent printer to respond at startup time.
2. You still don't realize that your printer is starting fine after 2 minutes.
3. Stop having Photonic3d open the com port that the printer wants to use at startup time.

If you are saying number 1 or number 2, then I agree. I wouldn't want to wait that long for a non-existent printer either. So, I'm asking you(in #313) to wait less time for your non-existent printer to respond by decreasing the timeout period.

I won't be doing number 3 because it means you could start six different printers using the same com port. Allowing something like that would put the functionality of Photonic3d in a fairly questionable position. Because then, instead of people asking me to do something silly(like ignore the comport), they would be asking me to do something reasonable by saying that Photonic 3d should reserve com ports when a printer starts.

Momboz52, 2017-04-07:
Hi Wes
Sorry for not being enough clear in my proposal.
First of all this could be a proposal for people investigating Photonic3D software and I don't think this is for normal use.
I'll try to say it again.
Let's assume that we are talking about a printer on COM2 as an example and we don’t have any printer connected to that COM2 port.
I would like to tell Photonic3D that I have a printer connected, although that's not true. And I would like Photonic3D taking a kind of default printer setting and print on that COM2 port normally, even though there is no printer connected to it. A kind of default control characters.
I would like to use this default control byte (or bytes) to do some synchronization with some external hardware or functions.
This has nothing to do with the duration of the timeouts in Photonic3D.
Best wishes / Mohamed

WesGilster, 2017-04-08:
Photonic3d doesn't stop you from sending whatever you want to the COMx port.

Momboz52, 2017-04-08:
That's fine, Wes.
Thank you.
I am not sure whether I did understand correctly the features in Photonic3D.
Could you, please, help me and tell how I could configure a Photonic3D printer that sends, at each layer change, i.e. at each PNG file exposure, the byte, let's say "A", on the configured motor serial port beside sending the PNG file on the simulated display.
This is the configuration I am trying to reach.
Here the Photonic3D printer I am thinking about:
[image]
I hope I am clear in what I am asking for and many thanks in advance.
BRs / Mohamed

WesGilster, 2017-04-08:
You seem to be asking for something clear, reasonable and straightforward. Let me know if this helps:
[image]

Momboz52, 2017-04-08:
Hi Wes
I have tested and this seems not to solve my issue.
I configured this as you proposed and after 2 minutes wait time. I have noticed that I got the string G91 on the motor serial port.
then I did start a job and I have been watching both outputs: the motor serial port and the projector serial port.
I could see just the first PNG image of my print job and nothing else. Exactly if Ph3D would be waiting for a feedback from the motor serial port.
Any view on that?

Momboz52, 2017-04-08:
Dear Wes
I think I have got it, but the string G91 during start of the printer...
Any view on that?

WesGilster, 2017-04-09:
The G91 is sent by the startup process to ensure a printer is available. It waits 2 minutes(or whatever is the configured time) for a response after which nothing comes and the startup process completes.

WesGilster, 2017-06-28:
No bug here. If you believe there is an issue. Feel free to reopen.

closed 2017-06-28 by WesGilster