area515/Photonic3D #319: Failded Couldn't unpack new job.........

Jolly2015, 2017-06-05:
I always got this error massage if the .cws file a bit huge,at the outset, I thought maybe SD card capacity it's not enough(8G). and then I replacement a 16G SD card,now it can unpack 185Mb,but when I print another files without restart machine,it will show the error status again(but can be fixed by restart),it is a bug?
below is some of information. 
Hardware:Raspberry PI3
Image:2016-11-25-raspbian-jessie-lite.img
log err:
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
Jun 05, 2017 10:51:56 AM org.fourthline.cling.UpnpServiceImpl <init>
INFO: >>> Starting UPnP service...
Jun 05, 2017 10:51:56 AM org.fourthline.cling.UpnpServiceImpl <init>
INFO: Using configuration: org.fourthline.cling.DefaultUpnpServiceConfiguration
Jun 05, 2017 10:51:56 AM org.fourthline.cling.transport.RouterImpl <init>
INFO: Creating Router: org.fourthline.cling.transport.RouterImpl
Jun 05, 2017 10:51:56 AM org.fourthline.cling.transport.impl.MulticastReceiverImpl init
INFO: Creating wildcard socket (for receiving multicast datagrams) on port: 1900
Jun 05, 2017 10:51:56 AM org.fourthline.cling.transport.impl.MulticastReceiverImpl init
INFO: Joining multicast group: /239.255.255.250:1900 on network interface: wlan0
Jun 05, 2017 10:51:56 AM org.fourthline.cling.transport.impl.StreamServerImpl init
INFO: Created server (for receiving TCP streams) on: /192.168.1.106:5001
Jun 05, 2017 10:51:56 AM org.fourthline.cling.transport.impl.DatagramIOImpl init
INFO: Creating bound socket (for datagram input/output) on: /192.168.1.106
Jun 05, 2017 10:51:56 AM org.fourthline.cling.UpnpServiceImpl <init>
INFO: <<< UPnP service started successfully

WesGilster, 2017-06-05:
You'll need to click the "download" button on any photonic3d page and upload the file to this bug report. Most likely the reason it works after a reboot is because all of the slice files have been cleared out of the temp dir. We've talked about some ways of making this process a bit more efficient. 
1. clear the tmp directory more often...
2. only extract the currently shown image.

WesGilster, 2017-06-11:
I guess another alternative would be to get a larger sd card.

Jolly2015, 2017-06-12:
@WesGilster  Yeah, it's not a big problem to me, and I will get a larger SD card.I just would like to see  Phonotic 3D system get more and more profect.

WesGilster, 2017-06-12:
We do as well! Which is why we'll leave this open as a feature request.