area515/Photonic3D #321: [Discussion] replace pi4j with diozero

kloknibor, 2017-06-09:
hi,

I just wanted to discuss if it would be a good idea to replace pi4j with diozero. Diozero seems to be more widely supported by different single boards computer and the performance of diozero seems higher. 

I'm looking into running Photonic3D on an asus Tinkerboard which is supported by Diozero already, probably pi4j will work as well but with some modding most likely.

Let me know your thoughts :)

Kind regards,

Robin

WesGilster, 2017-06-09:
I'm not really attached to pi4j. I'm not sure anyone is really using the 
functionality from pi4j anyway...

Thanks,
Wes G.

kloknibor, 2017-06-14:
in that case I might at in Diozero soon since I will have some usecases for it :)

WesGilster, 2017-06-15:
Interesting. Interested to see what those use cases look like!