area515/Photonic3D #322: : Failed Couldn't unpack new job

JULIOSOLER00, 2017-06-12:
Status: Failed Couldn't unpack new job:/root/uploaddir/Anillos 85.cws into working directory:/

WesGilster, 2017-06-22:
There can be many reasons for this, and it's difficult to determine without the a download diagnostic from Photonic3D. First most common issues are:
1. SDCard ran out of space because your zip is too large.
2. You don't have permissions to working directory.

WesGilster, 2017-06-28:
I'll assume that one of these two solutions solved your issue since I haven't gotten a download diagnostic yet.

JULIOSOLER00, 2017-07-04:
Sorry for replying too late, I could not solve that problem. My machine is KLD-1260 and has a memory of 8 gygabyte.

The file to upload is 190 mb

WesGilster, 2017-09-04:
You've run out of space on your disk.