area515/Photonic3D #323: [BUG] : Special characters not handled by Gcode

kloknibor, 2017-06-17:
Hi,

I need to send the following command : M42 P# S0

When I call the executeGcode API it forwards only this : 

2017-06-17 11:15:52,953 [qtp839998248-72] Write: M42 P
2017-06-17 11:15:52,953 [qtp839998248-72] Printer received:M42 P
2017-06-17 11:15:52,953 [qtp839998248-72] lineRead: ok

I probably need to encode or escape it :) Will look into that! But would be need if the API did that itself. I will take a look at that as well but any input is welcome!

WesGilster, 2017-06-19:
I just tried that, and it seems to work find without escaping.

kloknibor, 2017-06-19:
I misunderstood the request and the # is mainly a place holder but for me it really ends up returning as shown... this was from the eclipse environment though... On what platform did you test? Can that matter anyway?

closed 2017-06-19 by kloknibor

WesGilster, 2017-06-19:
parsing isn't dependent on the OS, so it really shouldn't matter.