area515/Photonic3D #324: [BUG] Windows config.properties hasn't got updaterepo

kloknibor, 2017-06-26:
Hi,

I see in start.sh we are looking for updaterepo in CONFIG_PROPS="${HOME}/3dPrinters/config.properties" but in %HOMEPATH%/3dPrinters/config.properties there is no updateRepo entry visable althoug this file does exist and for example contains : performedOneTimeInstall=true

For the updating start.bat to work it would be good to have updateRepo also available in the windows config.properties :)! 

I must mention I haven't checked if this is working correctly on linux! Maybe we need to fix both.

For now I will use printerProfileRepo... ;)!

WesGilster, 2017-06-28:
First off, I don't understand why there is any reason to have both a printerProfileRepo and updateRepo. Not sure if that was necessary.

As to your question about config.properties in the 3dPrinters directory, there is a good reason to do that. In that directory are the user specific settings that exist for any given user. The more commonly known settings are in the CWH root directory. What is important to note is that those settings will get overwritten on every install of CWH. the user specific settings will not get overwritten when CWH is upgraded and installed.

kloknibor, 2017-07-10:
Allright I agree, printerProfileRepo is just a bit confusing. I think it is good to have the repo in config.properties but maybe just call it simply "Repo" and we then get all our stuff from there!