area515/Photonic3D #325: cwhservice not starting in newest OS

kloknibor, 2017-07-06:
Photonic doesn't seem to start on boot anymore for me. So I tried to start the service manually and got this error log : 

pi@lchr:/opt/cwh $ sudo service cwhservice start
Job for cwhservice.service failed. See 'systemctl status cwhservice.service' and 'journalctl -xn' for details.
pi@lchr:/opt/cwh $ systemctl status cwhservice.service
● cwhservice.service - LSB: Starts the CWH
   Loaded: loaded (/etc/init.d/cwhservice)
   Active: failed (Result: exit-code) since Thu 2017-07-06 09:36:21 UTC; 25s ago
  Process: 5079 ExecStart=/etc/init.d/cwhservice start (code=exited, status=203/EXEC)

running the start command manually results in this : 
pi@lchr:/etc/init.d $ sudo /opt/cwh/start.sh > /opt/cwh/log.scrout 2> /opt/cwh/log.screrr
-bash: /opt/cwh/log.scrout: Permission denied

any idea what is going on... With sudo I should have permission I would think?

WesGilster, 2017-07-06:
Interesting. Apparently you don't. I think we need to readdress how we are setting the permissions when we initially setup the cwhdirectory. Did you just download the latest Raspbian?

kloknibor, 2017-07-10:
I did use the latest Raspbian. The latest raspbian image on our website still works though! I ended up using this and modifying it to my needs...

I also tried changing the comand to just /opt/cwh/start,sh this doesn't work to start the service but manually this code does run! To make it even stranger ;)!

WesGilster, 2017-07-13:
I think it's been over a year since I've tested this and it works great. I can't reproduce your issue. Your going to need to give me very precise steps. This is what I did:
1. Download Raspbian and burn the image.
2. Enable the SSH daemon in the desktop(since apparently they don't support it by default anymore)
3. SSH to the Raspberry Pi
4. Run the steps on page: https://wiki.photonic3d.com/doku.php?id=linux
5. Run sudo reboot
6. Browse to Photonic3D.

closed 2017-07-16 by kloknibor