area515/Photonic3D #326: [BUG] Windows printing screen doesn't always open in foreground

kloknibor, 2017-07-10:
Hi,

It seems that the java window (the printing screen) is sometimes minimized as soon as it is opens. Because of this the printing screen won't be displayed untill the java icon is clicked to open up the screen.

So we need to find a way to make sure the window opens correctly and in the foreground on windows. I'll take a quick look now but maybe @WesGilster will have a better idea to resolve this :)!

kloknibor, 2017-07-10:
So I just looked through the code and found out you already set always on top... Not sure what goes wrong : 

```
                refreshFrame.setAlwaysOnTop(true);
		refreshFrame.setUndecorated(true);
		refreshFrame.setExtendedState(JFrame.MAXIMIZED_BOTH);
```

kloknibor, 2017-07-11:
All tags I could find in the Jdisplay API seems set... I guess the easiest thing to do is just checking if it is open after x seconds and if not re-open it. 

It even seems like it does open maximized but then minimizes itself for some weird reason... What are your toughts @WesGilster ?

WesGilster, 2017-07-15:
I can't actually reproduce this. Is this only a problem when using /Display1 or does it happen with /Display0 as well?

WesGilster, 2017-07-15:
I think I get what you are talking about now. When you click on display0, display1 minimizes. Clicking the window again refocuses the window.

WesGilster, 2017-07-15:
Turning off full screen support fixes this bug. Looks like I can make that an option to utilize full screen mode or not.

WesGilster, 2017-07-15:
Added a full screen option that fixes this bug now...
[image]

kloknibor, 2017-07-16:
HI Wes! 

Cool thanks we will try testing this! I had a hard time reproducing this myself on a single screen as well but on a double screen machine it happend 80% of the time. 

I'll let you know if it works!