area515/Photonic3D #327: [Feature request] Generic pixel dimming

kloknibor, 2017-07-10:
A cool feature would be too be able to use generic pixel dimming. 

Larger exposure area's generally need less curing time than smaller exposure area's. To correct for this and optmizing the printing process it would cool to implement the following option : 

First we find out what the smallest group of pixels is. With this data we can determine the what the longest needed exposure time is for the current print. So the small pixel groups are 100% white 0% black.

After this we need to determine the greyscale for the larger exposure area's. To determine this we might want to use something like a slicing calculator since I'm not sure this is lineair. So how larger an area how more grey instead of white it will be.

Let me know what you think about this :)!

kloknibor, 2017-07-10:
possibly we might want a smarter implementation of what NanoDLP has : 
[image]

#Yes shamelessly stolen from their website :P

WesGilster, 2017-07-11:
Check out line 884 of:
org.area515.resinprinter.slice.ZSlicer.colorizePolygons(List<Face3d>, List<Integer>)

Calculating the size of the polygons is trivial, then you simply assign a color value to each polygon and away you go...

kloknibor, 2017-07-11:
@WesGilster  I took a long look at this but it's still a bit hocuspocus for me... I understand we seem to make a lists with the different area's but that's where my understanding ends... Sorry! Can you give me some more pointers?

WesGilster, 2017-07-12:
Certainly. 

A Zslicer slices a 3d figure. When a zslicer slices a cube, it produces square. When a zslicer slices a sphere, it produces a circle. When it slices a ball and pair of dice, it produces a circle and 2 squares in a single slice. In each of these situations you are left with 0 or many closed polygons(circles, squares and other complex loops). In this Zslicer, a polygon is represented as a simple group of connected lines: List<Line3d>. So to extend this a bit further, you'd represent a group of polygons as: List<List<Line3d>>. Notice that the type of variable of completedFillInLoops is exactly that: List<List<Line3d>>

So now, if you were to checkout line 884 of the colorizePolygons, you'd fine that we are simply looping through all of the polgons with: for (List<Line3d> currentPolygon : completedFillInLoops)

So your first step is to write a simple private method that finds the square area of "currentPolygon" and assign it a grayscale as you suggested. Then associate the scanlines that result to that grayscale.