area515/Photonic3D #329: Gradlew doesn't push to github anymore

kloknibor, 2017-07-16:
Hi,

I'm just wondering if I'm the only one with this issue currently... I tried to do a fullRelease from eclipse but it ends in an error at github release on all different repo's I manage, also tried a new personal acces token without any luck! I shouldn't have changed anything weird, only stuff in the resources folder so this shouldn't cause any java issues.

Here is the gradlew fullRelease --info command, let me know if debug helps more : 

C:\Users\Robin\Documents\GitHub\Cerin\Photonic3D-Dev\host>gradlew fullRelease --info
Initialized native services in: C:\Users\Robin\.gradle\native
Connected to daemon DaemonInfo{pid=3520, address=[70c10fbb-7ef8-4cbd-a9e4-09fb29a50461 port:52181, addresses:[/127.0.0.1, /0:0:0:0:0:0:0:1]], idle=true, lastBusy=1500221218599, context=DefaultDaemonContext[uid=03c2fc98-b914-44f4-982f-686fac0dff88,javaHome=C:\Program Files\Java\jdk1.8.0_121,daemonRegistryDir=C:\Users\Robin\.gradle\daemon,pid=3520,idleTimeout=10800000,daemonOpts=-XX:MaxPermSize=256m,-XX:+HeapDumpOnOutOfMemoryError,-Xmx1024m,-Dfile.encoding=windows-1252,-Duser.country=US,-Duser.language=en,-Duser.variant]}. Dispatching request Build{id=cf8426c7-7761-4cad-9889-f4ed974b3a3f.1, currentDir=C:\Users\Robin\Documents\GitHub\Cerin\Photonic3D-Dev\host}.
Received result org.gradle.launcher.daemon.protocol.BuildStarted@5454d35e from daemon DaemonInfo{pid=3520, address=[70c10fbb-7ef8-4cbd-a9e4-09fb29a50461 port:52181, addresses:[/127.0.0.1, /0:0:0:0:0:0:0:1]], idle=true, lastBusy=1500221218599, context=DefaultDaemonContext[uid=03c2fc98-b914-44f4-982f-686fac0dff88,javaHome=C:\Program Files\Java\jdk1.8.0_121,daemonRegistryDir=C:\Users\Robin\.gradle\daemon,pid=3520,idleTimeout=10800000,daemonOpts=-XX:MaxPermSize=256m,-XX:+HeapDumpOnOutOfMemoryError,-Xmx1024m,-Dfile.encoding=windows-1252,-Duser.country=US,-Duser.language=en,-Duser.variant]} (build should be starting).
The client will now receive all logging from the daemon (pid: 3520). The daemon log file: C:\Users\Robin\.gradle\daemon\3.0\daemon-3520.out.log
Starting 4th build in daemon [uptime: 3 mins 52.863 secs, performance: 100%, GC rate: 0.16/s, tenured heap usage: 4% of 716.2 MB]
Executing build with daemon context: DefaultDaemonContext[uid=03c2fc98-b914-44f4-982f-686fac0dff88,javaHome=C:\Program Files\Java\jdk1.8.0_121,daemonRegistryDir=C:\Users\Robin\.gradle\daemon,pid=3520,idleTimeout=10800000,daemonOpts=-XX:MaxPermSize=256m,-XX:+HeapDumpOnOutOfMemoryError,-Xmx1024m,-Dfile.encoding=windows-1252,-Duser.country=US,-Duser.language=en,-Duser.variant]
Starting Build
Settings evaluated using settings file 'C:\Users\Robin\Documents\GitHub\Cerin\Photonic3D-Dev\host\settings.gradle'.
Projects loaded. Root project using build file 'C:\Users\Robin\Documents\GitHub\Cerin\Photonic3D-Dev\host\build.gradle'.
Included projects: [root project 'photonic3d']
Evaluating root project 'photonic3d' using build file 'C:\Users\Robin\Documents\GitHub\Cerin\Photonic3D-Dev\host\build.gradle'.
------------------------------------------------------------------------
Detecting the operating system and CPU architecture
------------------------------------------------------------------------
os.detected.name=windows
os.detected.arch=x86_64
os.detected.classifier=windows-x86_64
All projects evaluated.
Selected primary task 'fullRelease' from project :
Tasks to be executed: [task ':setupRemote', task ':setupRelease', task ':compileJava', task ':processResources', task ':classes', task ':jar', task ':startScripts', task ':compileTestJava', task ':processTestResources', task ':testClasses', task ':testJar', task ':installDist', task ':repoVersion', task ':preDist', task ':BuildCWH', task ':dist', task ':releaseDist', task ':githubRelease', task ':fullRelease']
In-memory cache of C:\Users\Robin\Documents\GitHub\Cerin\Photonic3D-Dev\host\.gradle\3.0\taskArtifacts\fileHashes.bin: Size{1042}, CacheStats{hitCount=1268, missCount=1042, loadSuccessCount=0, loadExceptionCount=0, totalLoadTime=0, evictionCount=0}
In-memory cache of C:\Users\Robin\Documents\GitHub\Cerin\Photonic3D-Dev\host\.gradle\3.0\taskArtifacts\treeSnapshots.bin: Size{0}, CacheStats{hitCount=0, missCount=0, loadSuccessCount=0, loadExceptionCount=0, totalLoadTime=0, evictionCount=0}
In-memory cache of C:\Users\Robin\Documents\GitHub\Cerin\Photonic3D-Dev\host\.gradle\3.0\taskArtifacts\treeSnapshotUsage.bin: Size{0}, CacheStats{hitCount=0, missCount=0, loadSuccessCount=0, loadExceptionCount=0, totalLoadTime=0, evictionCount=0}
In-memory cache of C:\Users\Robin\Documents\GitHub\Cerin\Photonic3D-Dev\host\.gradle\3.0\taskArtifacts\fileSnapshotsToTreeSnapshotsIndex.bin: Size{9}, CacheStats{hitCount=0, missCount=9, loadSuccessCount=0, loadExceptionCount=0, totalLoadTime=0, evictionCount=0}
In-memory cache of C:\Users\Robin\Documents\GitHub\Cerin\Photonic3D-Dev\host\.gradle\3.0\taskArtifacts\fileSnapshots.bin: Size{37}, CacheStats{hitCount=0, missCount=28, loadSuccessCount=0, loadExceptionCount=0, totalLoadTime=0, evictionCount=0}
In-memory cache of C:\Users\Robin\Documents\GitHub\Cerin\Photonic3D-Dev\host\.gradle\3.0\taskArtifacts\taskArtifacts.bin: Size{11}, CacheStats{hitCount=0, missCount=11, loadSuccessCount=0, loadExceptionCount=0, totalLoadTime=0, evictionCount=0}
:setupRemote (Thread[Daemon worker,5,main]) started.
:setupRemote
Executing task ':setupRemote' (up-to-date check took 0.0 secs) due to:
  Task has not declared any outputs.
Releasing to: [https://github.com/Cerin-Photonic3D/Photonic3D-Dev.git]
Repo Owner: Cerin-Photonic3D / Repo Name: Photonic3D-Dev
:setupRemote (Thread[Daemon worker,5,main]) completed. Took 0.003 secs.
:setupRelease (Thread[Daemon worker,5,main]) started.
:setupRelease
Executing task ':setupRelease' (up-to-date check took 0.0 secs) due to:
  Task has not declared any outputs.
----------------------------------------------
Description of current commit: test
UNCOMMITTED STAGED CHANGES: org.ajoberstar.grgit.Status$Changes(added:[], modified:[], removed:[])
UNCOMMITTED UNSTAGED CHANGES: org.ajoberstar.grgit.Status$Changes(added:[host/photonic-Xtreme-1.75.zip, host/photonic-test.zip], modified:[], removed:[])
----------------------------------------------
Dry-Run Git Push to origin:
The following authentication options are allowed (though they may not be available): [SSHAGENT, PAGEANT, INTERACTIVE, HARDCODED]
using hardcoded credentials provided programmatically
----------------------------------------------
Check the above git push dry run to ensure that your push will succeed!
[ant:input] Press <return> to continue...
> Building 5% > :setupRelease
Recommended development release tag: Cerin-Photonic3D-1d5cc26
[ant:input] Enter the tag to create for this release >
> Building 5% > :setupReleasetest2
Added tag: test2
:setupRelease (Thread[Daemon worker,5,main]) completed. Took 8.104 secs.
:compileJava (Thread[Daemon worker,5,main]) started.
:compileJava
Skipping task ':compileJava' as it is up-to-date (took 0.31 secs).
:compileJava UP-TO-DATE
:compileJava (Thread[Daemon worker,5,main]) completed. Took 0.316 secs.
:processResources (Thread[Daemon worker,5,main]) started.
:processResources
Skipping task ':processResources' as it is up-to-date (took 0.006 secs).
:processResources UP-TO-DATE
[... 68 lines cut ...]
:githubRelease (Thread[Daemon worker,5,main]) started.
:githubRelease
Executing task ':githubRelease' (up-to-date check took 0.0 secs) due to:
  Task has not declared any outputs.
I/O exception (java.net.SocketException) caught when processing request to {s}->https://uploads.github.com: Connection reset by peer: socket write error
Retrying request to {s}->https://uploads.github.com
:githubRelease FAILED
:githubRelease (Thread[Daemon worker,5,main]) completed. Took 2 mins 0.901 secs.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':githubRelease'.
> org.apache.http.client.ClientProtocolException

* Try:
Run with --stacktrace option to get the stack trace. Run with --debug option to get more log output.

BUILD FAILED

Total time: 2 mins 15.843 secs
Stopped 0 compiler daemon(s).
Received result Failure[value=org.gradle.initialization.ReportedException: org.gradle.internal.exceptions.LocationAwareException: Execution failed for task ':githubRelease'.] from daemon DaemonInfo{pid=3520, address=[70c10fbb-7ef8-4cbd-a9e4-09fb29a50461 port:52181, addresses:[/127.0.0.1, /0:0:0:0:0:0:0:1]], idle=true, lastBusy=1500221218599, context=DefaultDaemonContext[uid=03c2fc98-b914-44f4-982f-686fac0dff88,javaHome=C:\Program Files\Java\jdk1.8.0_121,daemonRegistryDir=C:\Users\Robin\.gradle\daemon,pid=3520,idleTimeout=10800000,daemonOpts=-XX:MaxPermSize=256m,-XX:+HeapDumpOnOutOfMemoryError,-Xmx1024m,-Dfile.encoding=windows-1252,-Duser.country=US,-Duser.language=en,-Duser.variant]} (build should be done).

maybe it's just github having some hickups but status.github.com reports fully operational!

kloknibor, 2017-07-16:
here is the debug, also rebooted my pc :P Sorry for spamming all this information but I basically am not sure what happens with this...  

18:21:03.198 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Waiting to acquire shared lock on daemon addresses registry.
18:21:03.198 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Lock acquired.
18:21:03.199 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Releasing lock on daemon addresses registry.
18:21:13.196 [DEBUG] [org.gradle.launcher.daemon.server.Daemon] DaemonExpirationPeriodicCheck running
18:21:13.196 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Waiting to acquire shared lock on daemon addresses registry.
18:21:13.196 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Lock acquired.
18:21:13.197 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Releasing lock on daemon addresses registry.
18:21:23.196 [DEBUG] [org.gradle.launcher.daemon.server.Daemon] DaemonExpirationPeriodicCheck running
18:21:23.196 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Waiting to acquire shared lock on daemon addresses registry.
18:21:23.197 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Lock acquired.
18:21:23.197 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Releasing lock on daemon addresses registry.
18:21:33.196 [DEBUG] [org.gradle.launcher.daemon.server.Daemon] DaemonExpirationPeriodicCheck running
18:21:33.196 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Waiting to acquire shared lock on daemon addresses registry.
18:21:33.197 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Lock acquired.
18:21:33.197 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Releasing lock on daemon addresses registry.
18:21:43.196 [DEBUG] [org.gradle.launcher.daemon.server.Daemon] DaemonExpirationPeriodicCheck running
18:21:43.197 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Waiting to acquire shared lock on daemon addresses registry.
18:21:43.197 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Lock acquired.
18:21:43.198 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Releasing lock on daemon addresses registry.
18:21:53.196 [DEBUG] [org.gradle.launcher.daemon.server.Daemon] DaemonExpirationPeriodicCheck running
18:21:53.196 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Waiting to acquire shared lock on daemon addresses registry.
18:21:53.196 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Lock acquired.
18:21:53.197 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Releasing lock on daemon addresses registry.
18:22:03.196 [DEBUG] [org.gradle.launcher.daemon.server.Daemon] DaemonExpirationPeriodicCheck running
18:22:03.196 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Waiting to acquire shared lock on daemon addresses registry.
18:22:03.197 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Lock acquired.
18:22:03.197 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Releasing lock on daemon addresses registry.
18:22:13.197 [DEBUG] [org.gradle.launcher.daemon.server.Daemon] DaemonExpirationPeriodicCheck running
18:22:13.198 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Waiting to acquire shared lock on daemon addresses registry.
18:22:13.198 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Lock acquired.
18:22:13.199 [DEBUG] [org.gradle.cache.internal.DefaultFileLockManager] Releasing lock on daemon addresses registry.
18:22:18.551 [DEBUG] [org.apache.http.impl.conn.DefaultClientConnection] Connection 0.0.0.0:52923<->192.30.253.98:443 closed
18:22:18.551 [DEBUG] [org.apache.http.impl.client.DefaultHttpClient] Closing the connection.
18:22:18.551 [DEBUG] [org.apache.http.impl.conn.DefaultClientConnection] Connection 0.0.0.0:52923<->192.30.253.98:443 closed
18:22:18.551 [INFO] [org.apache.http.impl.client.DefaultHttpClient] I/O exception (java.net.SocketException) caught when processing request to {s}->https://uploads.github.com: Connection reset by peer: socket write error
18:22:18.552 [DEBUG] [org.apache.http.impl.client.DefaultHttpClient] Connection reset by peer: socket write error
java.net.SocketException: Connection reset by peer: socket write error
        at java.net.SocketOutputStream.socketWrite0(Native Method)
        at java.net.SocketOutputStream.socketWrite(SocketOutputStream.java:111)
        at java.net.SocketOutputStream.write(SocketOutputStream.java:155)
        at sun.security.ssl.OutputRecord.writeBuffer(OutputRecord.java:431)
        at sun.security.ssl.OutputRecord.write(OutputRecord.java:417)
        at sun.security.ssl.SSLSocketImpl.writeRecordInternal(SSLSocketImpl.java:876)
        at sun.security.ssl.SSLSocketImpl.writeRecord(SSLSocketImpl.java:847)
        at sun.security.ssl.AppOutputStream.write(AppOutputStream.java:123)
        at org.apache.http.impl.io.AbstractSessionOutputBuffer.write(AbstractSessionOutputBuffer.java:181)
        at org.apache.http.impl.io.ContentLengthOutputStream.write(ContentLengthOutputStream.java:115)
        at org.apache.http.entity.InputStreamEntity.writeTo(InputStreamEntity.java:142)
        at org.apache.http.entity.HttpEntityWrapper.writeTo(HttpEntityWrapper.java:89)
        at org.apache.http.impl.client.EntityEnclosingRequestWrapper$EntityWrapper.writeTo(EntityEnclosingRequestWrapper.java:108)
[... cut ...] =C:\Users\Robin\.gradle\daemon,pid=3520,idleTimeout=10800000,daemonOpts=-XX:MaxPermSize=256m,-XX:+HeapDumpOnOutOfMemoryError,-Xmx1024m,-Dfile.encoding=windows-1252,-Duser.country=US,-Duser.language=en,-Duser.variant]} (build should be done).
18:22:18.860 [DEBUG] [org.gradle.launcher.daemon.client.DaemonClientConnection] thread 1: dispatching class org.gradle.launcher.daemon.protocol.Finished
18:22:18.861 [DEBUG] [org.gradle.launcher.daemon.client.DaemonClientConnection] thread 1: connection stop

jmkao, 2017-07-16:
Strange, it does look like a github side network error in the stack. I'll check it, give me a few hours to get to my laptop.

kloknibor, 2017-07-16:
@jmkao I thought the same since the socket gets closed and it's on all my repo's ... But it's still at 100% availability.

Thanks in advance! I'll retry it in a bit as well ;)!

WesGilster, 2017-07-16:
I haven't built from gradlew for quite some time due to this:
https://github.com/area515/Photonic3D/issues/275

However, then again. I really haven't needed to since I've used Travis for all my builds. 

Speaking of builds... Aren't we long overdue for a prod release?

jmkao, 2017-07-16:
Just did it (on a Mac, don't have a Windows laptop with me), and it worked fine. I notice that upload.github.com uses a fastly CDN, so it's possible that there is something wrong with the European gateway that doesn't affect other egress points, and thus doesn't reflect in the uptime dashboard.

kloknibor, 2017-07-16:
@WesGilster  yes I agree we should do a new prod release!

@jmkao It seems that now after half a day it works again, I just rechecked github status which didn't give an error and also fastly didn't report any problems. So it stays weird but well it works again so I'll close it for now :)! Thanks for thinking with me!

kloknibor, 2017-07-16:
allright I spoke to early... it uploaded the earlier failed zip files but now fails again... I might just set-up Travis tomorrow as well :/

kloknibor, 2017-07-17:
ok it works again... Will close this for now!

closed 2017-07-17 by kloknibor