area515/Photonic3D #331: Semaphore CI-based RPi image builds not working

jmkao, 2017-07-18:
Semaphore CI based RPi image builds are failing at trivial steps like the `sudo modprobe loop` for some unknown reason. After a few hours of WTF, there is a viable way to do this in Travis and add in the capability to have triggered builds so that whenever a new prod release goes out to area515/Photonic3D, a new image will automatically be created. Working on that now.

WesGilster, 2017-07-18:
Awesome, now that's really cool.

WesGilster, 2017-07-19:
Looks like we have a legit bug with our image, where the desktop doesn't initialize... I know I remember you guys working through an XWindows problem on the Raspbian-Lite image a long time ago. Do you guys remember the fix? 
1. Is it possible that there isn't a user that logs in properly so the xenvironment isn't initialized? (I noticed bootstrapping is paused at the login prompt)
2. Is it possible that Raspbian-Lite doesn't have xwindows configured/installed?

Here are the users having troubles...
https://forum.photonic3d.com/t/could-not-start-printing-could-not-initialize-class-sun-awt-x11graphicsenvironment/183/19

jmkao, 2017-07-19:
Yeah, I'll tackle them both at once. A change to fix this now that I know that I need to check for X should be relatively simple with the build times amortized over getting the S3 deploy working again.

jmkao, 2017-08-02:
As of 1bff32ba3c3ff3847be47e9ff543aba4b8cbf929, Travis CI based building works, and the image has been validated to boot properly as well.

Still need to set up the trigger in area515/Photonic3D to rebuild the image whenever a new production build is merged.

WesGilster, 2017-08-03:
So are you going to setup production to build on any commit?

jmkao, 2017-08-04:
Yeah, or more specifically, any tagged commit that triggers the creation of a new release will also trigger the update of the image.

jmkao, 2017-08-05:
3ff6b16 should trigger image creations upon deployment now.

@WesGilster Is your fork stable enough for us to merge the last few months of enhancements into production?

WesGilster, 2017-08-05:
I think so. I just got that last changes put in for Robin and a couple of changes I needed to auto fix the aspect ratio(as a Customizer) on my own prints.

WesGilster, 2017-08-05:
You want to do the honors?

jmkao, 2017-08-05:
I think you need to do it since the PR is coming from your repo :)

WesGilster, 2017-08-06:
I must be missing something, I kindof figured the release would be automatically created from the merge as well. I still do the merge manually then?

jmkao, 2017-08-06:
You've already merged it. You need to create a tag like "1.0.16" with the description you want for the release, and then push it. The presence of the tag will trigger all the automagic action.

WesGilster, 2017-08-06:
I see, so the triggers are the same as it's always been. I thought you were saying the build was triggered from the merge. Works awesome thanks! This is really cool.

jmkao, 2017-08-10:
1.0.16 validated this is fully working.

closed 2017-08-10 by jmkao