area515/Photonic3D #336: How to make a stand alone dlp machine with tft screen, and if we can exclude the arduino and use only the raspberry pi

lalbe, 2017-08-20:
First, I'm looking for a way to use an RPI TFT screen so I can make use the printer without needing for an additional device and network.

How can I split the screen to show the gui with the TFT screen and can send the slices to throw the HDMI cable?

Second is there any way to be able to not use the Arduino and control the servos from the pi it self?

I think I can interpret the g code but it will require me to be able to pass it to script I will write, and not pass it throw the serial cable

WesGilster, 2017-08-20:
I'm not sure why my email didn't show up here...
Even though XWindows doesn't support the second screen on a Raspberry Pi, we actually do. Here is how:
https://wiki.photonic3d.com/doku.php?id=dual_raspberry_pi_screens

Here is the process for creating a new skin for use on your touch screen:
https://wiki.photonic3d.com/doku.php?id=supporting_new_skins

Yes you can control the servos yourself. In fact it already ships with the libraries to do this. Obviously you'll need to write the code yourself since we have no idea what controls and servos you happen to be using. Here is where we were talking about it:
https://forum.photonic3d.com/t/wiring-diagram-suitable-for-photonic-3d/228/2

I think I can interpret the g code but it will require me to be able to pass it to script I will write, and not pass it throw the serial cable
You could do this, but it's not necessary. Technically you can write your own language which is convenient for you since you will be in control of the parsing. I'll help in any way I can to get you through the code.

WesGilster, 2017-08-20:
We'll be getting the JavaDocs and Restful API docs generating onbuild fairly soon.

WesGilster, 2017-08-22:
...and here is how another user figured it out:
https://forum.photonic3d.com/t/printflow3d-download/225/42

WesGilster, 2017-12-18:
So I've put the final nail in the coffin of this bug by recently allowing the ability to control the stepper motors directly as you requested. If you want to install the printer it's Called "Direct stepper control". If you want to create your own printer from scratch, you'll need to use the "Javascript Control" print driver as shown in the image below. Here is a little demo program for how you can control the steppers from java: https://github.com/Pi4J/pi4j/blob/master/pi4j-example/src/main/java/StepperMotorGpioExample.java

[image]