area515/Photonic3D #341: Automatic on / off the projector.

Ustas79U, 2017-09-13:
I use Photonics 3D on my printer with a DLP projector. I control the projector by RS232.
Manually turning the projector on and off is working OK.

[image]

Can it be possible to automatically turn on the projector when printing starts, and turn off when it finish?
Many thanks.

WesGilster, 2017-09-14:
Yep you can do that. If your projector isn't already supported, you can add your printer hex codes in your config.properties file under the key:
hexCodeBasedProjectors

As you can see, we already have the following existing projectors supported by default:
[  
   {  
      "name":"Optoma (HD141x)",
      "defaultComPortSettings":{  
         "speed":9600,
         "databits":8,
         "parity":"None",
         "stopbits":"One"
      },
      "onHex":"7E303030302031",
      "offHex":"7E303030302032",
      "bulbHoursHex":"7E30303135302031",
      "bulbHoursResponseRegex":"(?s:OK(?:0|1)(\\\\d{4}.*))",
      "bulbHoursConversion":"ASCII"
   },
   {  
      "name":"Vivitek (D535)",
      "defaultComPortSettings":{  
         "speed":9600,
         "databits":8,
         "parity":"None",
         "stopbits":"One"
      },
      "onHex":"7E504E0D",
      "offHex":"7E50460D",
      "bulbHoursHex":"7E714C0D(function (){java.lang.Thread.sleep(100);})()",
      "bulbHoursResponseRegex":"[^\\\\d]*(\\\\d+)\\\\x0D?",
      "bulbHoursConversion":"ASCII",
      "detectionHex":"7E504E0D(function (){java.lang.Thread.sleep(1);})()7E71560D",
      "detectionResponseRegex":"(?s:.*DP3406.*\\\\x0D?)"
   },
   {  
      "name":"Acer",
      "defaultComPortSettings":{  
         "speed":9600,
         "databits":8,
         "parity":"None",
         "stopbits":"One"
      },
      "onHex":"4F4B4F4B4F4B4F4B4F4B0D",
      "offHex":"2A2030204952203030320D",
      "detectionHex":"2A2030204C616D70203F0D",
      "detectionResponseRegex":"(?s:.*?)(?i:Lamp) [10](?s:.*?)"
   },
   {  
      "name":"ViewSonic (pjd7820hd)",
      "defaultComPortSettings":{  
         "speed":115200,
         "databits":8,
         "parity":"None",
         "stopbits":"One"
      },
      "onHex":"0614000400341100005D",
      "offHex":"0614000400341101005E",
      "bulbHoursHex":"0714000500340000150163",
      "bulbHoursResponseRegex":"\\\\x05\\\\x14\\\\x00\\\\x06\\\\x00\\\\x00\\\\x00([\\\\x00-\\\\xFF]{2})\\\\x00\\\\x00[\\\\x00-\\\\xFF]",
      "bulbHoursConversion":"LittleEndian",
      "detectionHex":"071400050034000011005E",
      "detectionResponseRegex":"(?s:.*(?:\\\\x05\\\\x14\\\\x00\\\\x03\\\\x00\\\\x00\\\\x00[\\\\x00\\\\x01][\\\\x17\\\\x18]|\\\\x03\\\\x14\\\\x00\\\\x00\\\\x00\\\\x14)\\\\x0D?)"
   }
]

WesGilster, 2018-01-21:
Closing since functionality is already supported.

closed 2018-01-21 by WesGilster