area515/Photonic3D #342: Weird Java Error I can't get to the bottom of...  hopefully it's a quick nudge for one of the masters here

thetrebor, 2017-10-02:
CW is awful, so i've reverse engineered the file format and have a nice modern and not-crashy app generating CWS files because that's the only way to have this model printer work.
I thought I've slavishly copied everything and it indeed seems to work right through up to printing but then when you go to print the job fails immediately with some inscrutable (to me) java errors about an appending process failing.  Can you point me to what's making it mad?

[LogBundle (1).zip](https://github.com/area515/Photonic3D/files/1350683/LogBundle.1.zip)
[LogBundle (2).zip](https://github.com/area515/Photonic3D/files/1350684/LogBundle.2.zip)

[pallisade2 2.stl.cws.zip](https://github.com/area515/Photonic3D/files/1350685/pallisade2.2.stl.cws.zip)

WesGilster, 2017-10-03:
Looks like macs seem to create these odd looking directories with files in them that are garbage:
[image]

This confuses our search routines which attempt to find the '.png' files. In the dev version I've fixed the file search to ignore things that are both in the __MACOSX directory and start with a '.'

thetrebor, 2017-10-03:
Interesting.  tonight i'll purge those out of the .cws file and see if it behaves better.  Mac puke files are not just hidden but *super hidden* on a mac which meant i didn't see them when making the compressed file.  Do you know why it would only fail on .cws files with the mac puke and not plain .zip files with mac puke?  I've compressed files from the same computer to plain .zip with none of the extra .cws internal structure and it largely worked.  Is it because they're not in the manifest.xml?

Note that i've also totally given up on using creation workshop and haven't had any more trouble with the errors we were seeing before with old files not deleting and causing subsequent uploads to fail.  There may be a nagging bug in the way cws is parsed?

thetrebor, 2017-10-03:
okay, so i don't think the mac puke is the problem.

i did this:
zip -d 20mmTestCube_repaired.stl.cws __MACOSX .DS_Store
zip warning: name not matched: __MACOSX

zip error: Nothing to do! (20mmTestCube_repaired.stl.cws)
[ Trebor Roberts-MBP ~/Desktop ] zip -d 20mmTestCube_repaired.stl.cws .DS_Store
zip error: Nothing to do! (20mmTestCube_repaired.stl.cws)

Then to make sure there was no mac puke i unzipped the file on a unix host and examined the output and it was fine.
[ec2-user@ip-172-30-2-151 tmp]$ ls
0065aedf-d370-4cc7-a74b-2ae80ae2b05f.gcode  114.png  130.png  147.png  163.png  17.png   196.png                        2.png   46.png  62.png  79.png  95.png
0065aedf-d370-4cc7-a74b-2ae80ae2b05f.json   115.png  131.png  148.png  164.png  180.png  197.png                        30.png  47.png  63.png  7.png   96.png
0065aedf-d370-4cc7-a74b-2ae80ae2b05f.stl    116.png  132.png  149.png  165.png  181.png  198.png                        31.png  48.png  64.png  80.png  97.png
100.png                                     117.png  133.png  14.png   166.png  182.png  199.png                        32.png  49.png  65.png  81.png  98.png
101.png                                     118.png  134.png  150.png  167.png  183.png  19.png                         33.png  4.png   66.png  82.png  99.png
102.png                                     119.png  135.png  151.png  168.png  184.png  1.png                          34.png  50.png  67.png  83.png  9.png
103.png                                     11.png   136.png  152.png  169.png  185.png  200.png                        35.png  51.png  68.png  84.png  LCHR Firm Amber 100 Êm.slicing
104.png                                     120.png  137.png  153.png  16.png   186.png  20mmTestCube_repaired.stl.cws  36.png  52.png  69.png  85.png  manifest.xml
105.png                                     121.png  138.png  154.png  170.png  187.png  20.png                         37.png  53.png  6.png   86.png
106.png                                     122.png  139.png  155.png  171.png  188.png  21.png                         38.png  54.png  70.png  87.png
107.png                                     123.png  13.png   156.png  172.png  189.png  22.png                         39.png  55.png  71.png  88.png
108.png                                     124.png  140.png  157.png  173.png  18.png   23.png                         3.png   56.png  72.png  89.png
109.png                                     125.png  141.png  158.png  174.png  190.png  24.png                         40.png  57.png  73.png  8.png
10.png                                      126.png  142.png  159.png  175.png  191.png  25.png                         41.png  58.png  74.png  90.png
110.png                                     127.png  143.png  15.png   176.png  192.png  26.png                         42.png  59.png  75.png  91.png
111.png                                     128.png  144.png  160.png  177.png  193.png  27.png                         43.png  5.png   76.png  92.png
112.png                                     129.png  145.png  161.png  178.png  194.png  28.png                         44.png  60.png  77.png  93.png
113.png                                     12.png   146.png  162.png  179.png  195.png  29.png                         45.png  61.png  78.png  94.png

But i still get the error when trying to print it:
[LogBundle.zip](https://github.com/area515/Photonic3D/files/1354129/LogBundle.zip)

Example failing file:
https://www.dropbox.com/s/z67uy3nah2ybhte/20mmTestCube_repaired.stl.cws?dl=0

Example of interface seeming like everything is fine:
https://www.dropbox.com/s/abjgkhz4v30j1m7/Screenshot%202017-10-03%2018.45.56.png?dl=0

Example of interface being sad
https://www.dropbox.com/s/en066ruw1xv28ca/Screenshot%202017-10-03%2018.46.18.png?dl=0

I don't think it's the diacritic in the file name because i have examples of those working.  The only real difference i can see is that my generated file names are fairly long (the uuid looking things) and in other examples they're shorter.  i guess the other difference is that i didn't left pad the numbers with zeroes so they're not in slice order if that makes a difference.
I would totally debug this myself if i could interpret the debug output!  Sorry I can't RTFM on this.

thetrebor, 2017-10-03:
(hopefully it's not the zombified remains of previously failed print jobs causing the issue.  the interface into the lchr is limited, not sure how i would do in restoring it to factory state)

WesGilster, 2017-10-04:
I can assure you that those Mac files will be a problem for you if they aren't already.

Your current problem is this:
```
2017-10-03 22:39:21,751 qtp506207-136 ERROR An exception occurred processing Appender RollingFile java.lang.NullPointerException
	at org.apache.logging.log4j.core.appender.rolling.AbstractRolloverStrategy.getEligibleFiles(AbstractRolloverStrategy.java:107)
	at org.apache.logging.log4j.core.appender.rolling.AbstractRolloverStrategy.getEligibleFiles(AbstractRolloverStrategy.java:96)
	at org.apache.logging.log4j.core.appender.rolling.AbstractRolloverStrategy.getEligibleFiles(AbstractRolloverStrategy.java:89)
	at org.apache.logging.log4j.core.appender.rolling.DefaultRolloverStrategy.purgeAscending(DefaultRolloverStrategy.java:212)
	at org.apache.logging.log4j.core.appender.rolling.DefaultRolloverStrategy.purge(DefaultRolloverStrategy.java:199)
	at org.apache.logging.log4j.core.appender.rolling.DefaultRolloverStrategy.rollover(DefaultRolloverStrategy.java:326)
	at org.apache.logging.log4j.core.appender.rolling.RollingFileManager.rollover(RollingFileManager.java:310)
	at org.apache.logging.log4j.core.appender.rolling.RollingFileManager.rollover(RollingFileManager.java:227)
	at org.apache.logging.log4j.core.appender.rolling.RollingFileManager.checkRollover(RollingFileManager.java:207)
	at org.apache.logging.log4j.core.appender.RollingFileAppender.append(RollingFileAppender.java:267)
	at org.apache.logging.log4j.core.config.AppenderControl.tryCallAppender(AppenderControl.java:156)
	at org.apache.logging.log4j.core.config.AppenderControl.callAppender0(AppenderControl.java:129)
	at org.apache.logging.log4j.core.config.AppenderControl.callAppenderPreventRecursion(AppenderControl.java:120)
	at org.apache.logging.log4j.core.config.AppenderControl.callAppender(AppenderControl.java:84)
	at org.apache.logging.log4j.core.config.LoggerConfig.callAppenders(LoggerConfig.java:448)
	at org.apache.logging.log4j.core.config.LoggerConfig.processLogEvent(LoggerConfig.java:433)
	at org.apache.logging.log4j.core.config.LoggerConfig.log(LoggerConfig.java:417)
	at org.apache.logging.log4j.core.config.LoggerConfig.log(LoggerConfig.java:403)
	at org.apache.logging.log4j.core.config.AwaitCompletionReliabilityStrategy.log(AwaitCompletionReliabilityStrategy.java:63)
	at org.apache.logging.log4j.core.Logger.logMessage(Logger.java:146)
	at org.apache.logging.log4j.spi.AbstractLogger.logMessageSafely(AbstractLogger.java:2091)
	at org.apache.logging.log4j.spi.AbstractLogger.logMessage(AbstractLogger.java:2005)
	at org.apache.logging.log4j.spi.AbstractLogger.logIfEnabled(AbstractLogger.java:1876)
	at org.apache.logging.log4j.spi.AbstractLogger.info(AbstractLogger.java:1421)
	at org.area515.resinprinter.services.PrintableService.getPrintables(PrintableService.java:280)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.jboss.resteasy.core.MethodInjectorImpl.invoke(MethodInjectorImpl.java:137)
	at org.jboss.resteasy.core.ResourceMethodInvoker.invokeOnTarget(ResourceMethodInvoker.java:296)
	at org.jboss.resteasy.core.ResourceMethodInvoker.invoke(ResourceMethodInvoker.java:250)
	at org.jboss.resteasy.core.ResourceMethodInvoker.invoke(ResourceMethodInvoker.java:237)
	at org.jboss.resteasy.core.SynchronousDispatcher.invoke(SynchronousDispatcher.java:356)
	at org.jboss.resteasy.core.SynchronousDispatcher.invoke(SynchronousDispatcher.java:179)
	at org.jboss.resteasy.plugins.server.servlet.ServletContainerDispatcher.service(ServletContainerDispatcher.java:220)
	at org.jboss.resteasy.plugins.server.servlet.HttpServletDispatcher.service(HttpServletDispatcher.java:56)
	at org.jboss.resteasy.plugins.server.servlet.HttpServletDispatcher.service(HttpServletDispatcher.java:51)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:790)
	at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:835)
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1685)
	at org.eclipse.jetty.websocket.server.WebSocketUpgradeFilter.doFilter(WebSocketUpgradeFilter.java:225)
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1668)
	at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java:581)
	at org.eclipse.jetty.server.session.SessionHandler.doHandle(SessionHandler.java:224)
	at org.eclipse.jetty.server.handler.ContextHandler.doHandle(ContextHandler.java:1158)
	at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:511)
	at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:185)
	at org.eclipse.jetty.server.handler.ContextHandler.doScope(ContextHandler.java:1090)
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:141)
	at org.eclipse.jetty.server.handler.HandlerList.handle(HandlerList.java:52)
	at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:119)
	at org.eclipse.jetty.server.Server.handle(Server.java:517)
	at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:308)
	at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:242)
	at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:273)
[... cut ...]
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:654)
	at org.eclipse.jetty.util.thread.QueuedThreadPool$3.run(QueuedThreadPool.java:572)
	at java.lang.Thread.run(Thread.java:745)
```

Which is due to an old version of log4j that has a bug:
https://issues.apache.org/jira/browse/LOG4J2-1815

I fixed this a while back and we are currently using this version:
  compile 'org.apache.logging.log4j:log4j-slf4j-impl:2.8.2'
  compile 'org.apache.logging.log4j:log4j-api:2.8.2'
  compile 'org.apache.logging.log4j:log4j-core:2.8.2'

thetrebor, 2017-10-04:
i believe you on the mac puke.  I'll make sure none of that gets there henceforth.
apparently the decrepit version of photonic shipped with the LCHR is an issue and I don't have the ability to upgrade it apart from them.  I'll talk to them and see about updating it.  In the mean time is there a way i can work around it?  
You are a scholar and a gentleman, kind sir.

thetrebor, 2017-10-04:
Okay new data!
The thing that seems to be triggering the error is the gcode file.  I took things out of the zip one by one until the printer would respond.  So it appears that the gcode file itself is the thing that's causing issues.   I changed the file name of the gcode file and also tried nesting it deeper so it definitely has a parent directory but to no avail. 
THEN i tried printing a different file which was already on the machine and which had printed out perfectly previously but was a CWS file with an embedded gcode and it also crashed!

So GOOD NEWS! My code isn't broken.  Bad news! My printer is broken.

WesGilster, 2017-10-04:
Well, it would be much easier if they upgraded their Gradle build file. The lines I pasted above are all it takes. Their fork of the software is not well maintained or kept in sync with most of the fixes in this repo.

If you are going to do it, you'll need to SSH into your printer with user:
raspberry 
and password:
photocentic.

Then run this:
pi@photonic3d:/opt/cwh/lib $ ls -al log*
rename those from *.jar to *.old

Then add in the following log4j libraries from maven or wherever you want...
-rw-r--r-- 1 root root  228154 Oct  3 03:53 log4j-api-2.8.2.jar
-rw-r--r-- 1 root root 1407853 Oct  3 03:53 log4j-core-2.8.2.jar
-rw-r--r-- 1 root root   23124 Oct  3 03:53 log4j-slf4j-impl-2.8.2.jar

A bit of a small surgery, but you get the idea...

thetrebor, 2017-10-04:
Glorious. I'm also going to take them to task for not keeping up on the
fork.
Having SSH to the machine will make things so much easier. I'll let you
know how it goes.

thetrebor, 2017-10-07:
Just getting to this now.  The password "photocentic" or "photocentric" or "photocentric." or various combinations thereof doesn't seem to work.  tried u:pi p:raspberry as well... would there be any other permutations?  Googling around now but can't seem to find anything.

WesGilster, 2017-10-09:
I don't know man. It's possible they changed the default password and didn't check the code into github. Here is where they setup the pi for the first time. Checkout the "newpassword" variable in the script:

https://github.com/Photocentric3D/Photonic3D/blob/master/host/bin/rpi-setup.sh

thetrebor, 2017-10-09:
This helped a great deal, thank you.  the answer is user: *pi*  and
password: *photocentric*
I'm in and i'll try making your suggested changes.  Hopefully I can
contribute a bit to this project.  I'm a reasonably dangerous hack with UI
/ UX stuff.

thetrebor, 2017-10-09:
I am happy to report i've worked through all the bugs in my slicer and upgraded the jar files to a version that doesn't crash when doing log rotations.  Thank you very much for all your help!

closed 2017-10-09 by thetrebor

WesGilster, 2017-10-09:
Good deal, that wasn't as bad as I thought. I figured we'd have to work through a couple more issues before we got you going.

thetrebor, 2017-10-10:
Well i did, but your code is pretty readable, even though I don't know a
lick of java.  The thing that stymied me the longest was realizing you were
looking for image slice files starting at 0 rather than 1, which completed
the "off by one error" requirement of every project and allowed me to
progress :)

Good news is that I can now slice, bundle, and send files to PrintFlow from
a native desktop application that compiles into windows, osx, or linux
executables.  Depending how things go with Photocentric I'll pursue
releasing this to the community.