area515/Photonic3D #344: fail to install on linux

kisnoge, 2017-11-20:
Are anyone successes install photonic3d on linux system?
I get error message as below.
==========================message start================
sudo ./start.sh
Local Config: /root/3dPrinters/config.properties
mv: cannot stat ‘cwh-.*.zip’: No such file or directory
Checking for new version from Github Repo: area515/Creation-Workshop-Host
grep: build.number: No such file or directory
./start.sh: line 113: curl: command not found
Local Tag:
Network Tag:
Couldn't fetch version from GitHub, launching existing install.
Turning off screen saver and power saving
No protocol specified
xset:  unable to open display ":0.0"
No protocol specified
xset:  unable to open display ":0.0"
No protocol specified
xset:  unable to open display ":0.0"
Installing CWH as a service
cp: cannot stat ‘/opt/cwh/cwhservice’: No such file or directory
chmod: cannot access ‘/etc/init.d/cwhservice’: No such file or directory
update-rc.d: error: initscript does not exist: /etc/init.d/cwhservice
Determinging if one time install has occurred
grep: /root/3dPrinters/config.properties: No such file or directory
Starting printer host server
==========================message end================

How to solve the problem ?

WesGilster, 2017-11-20:
You need to install curl first and try again. What version of Linux do 
you have that doesn't have curl installed? Just curious.

kisnoge, 2017-11-20:
problem still here. the error message as below, and my system is ubuntu 16.04

==================error start==================
After this operation, 306 kB of additional disk space will be used.
Get:1 http://tw.archive.ubuntu.com/ubuntu xenial/universe amd64 dos2unix amd64 6.0.4-1 [68.7 kB]
Fetched 68.7 kB in 0s (462 kB/s) 
Selecting previously unselected package dos2unix.
(Reading database ... 173523 files and directories currently installed.)
Preparing to unpack .../dos2unix_6.0.4-1_amd64.deb ...
Unpacking dos2unix (6.0.4-1) ...
Processing triggers for man-db (2.7.5-1) ...
Setting up dos2unix (6.0.4-1) ...
W: --force-yes is deprecated, use one of the options starting with --allow instead.
grep: /etc/init.d/cwhservice: No such file or directory
chmod: cannot access '/etc/init.d/cwhservice': No such file or directory
Turning off screen saver and power saving
Installing CWH as a service
Determinging if one time install has occurred
grep: /home/pi/3dPrinters/config.properties: No such file or directory
Starting printer host server
===============error msg end==================================

WesGilster, 2017-11-21:
Could you describe what problem you are having?

WesGilster, 2017-11-22:
It does look as though we may have to start supporting upstart instead of the old fashioned init.d. I believed Ubuntu supported both, however it looks that's not true. Until then, you'll have to start Photonic manually instead of having it automatically start when your system starts.

kisnoge, 2017-11-23:
Finally I could running webpage, but 
1.It could not connect mega board when add new printer.
2.The jdk is also renew "jdk-8u102-linux-x64.tar.gz" is not working. so I change to jdk-8u154-linux-x64.tar.gz and use mirror site.
3.I found cwservice is not running, seems still something still wrong.
4.the system is run on Ubuntu 16.04 LTS. Would you like to try this to find excetlly problem?

WesGilster, 2017-11-23:
1.It could not connect mega board when add new printer.
Please post the diagnostic dump download and open a different issue, it has nothing to do with issue.

2.The jdk is also renew "jdk-8u102-linux-x64.tar.gz" is not working. so I change to jdk-8u154-linux-x64.tar.gz and use mirror site.
Photonic3D will work with both of those.

3.I found cwservice is not running, seems still something still wrong.
Read my previous post. I already mentioned in my previous post that you'll need to manually start Photonic3D instead of it starting by itself on bootup. This has no operational effect on Photonic3D itself.

4.the system is run on Ubuntu 16.04 LTS. May I have install manually step with Phoeonic3d?
Not sure what you mean here, but you've already successfully installed Photonic3D.

5. Is install Ubuntu 14.0x could solve these problem?
No, your version of Ubuntu is just fine.

kisnoge, 2017-11-23:
Thanks a lot for early reply. It's already solve most problem.
Latest question, how to manually start Photonic3D ?
I could do autostart at boot, if I can manually start Photonic3D.

WesGilster, 2017-11-23:
1. You can start it a million ways with Linux and you are already manually starting it with start.sh. Here are the 3 possible ways:
     a. Just call start.sh. That's what you are doing now.
     b. Install /opt/cwh/cwhservice into init.d. This is what Photonic3D already attempts to do by itself. Since your system doesn't have init.d, this isn't a great option. 
     C. Your best option is to use upstart: http://upstart.ubuntu.com/. Once you install Photonic3D with upstart, post your solution here and we'll add it to Photonic so others that don't have init.d can autostart Photonic3D as well.

2. Yes. The first display in Linux is generally numbered :0.0 under xWindows. If this is all too confusing, just choose "First available display". If you only have 1 display, it will pick that one.

DanielJoyce, 2017-12-15:
init.d is largely dead and upstart is deprecated. All the big distros are moving to systemd.

WesGilster, 2017-12-16:
Updated title to move to systemd.