area515/Photonic3D #345: Offline printing!

olegmorgun, 2017-12-20:
I'm very successful at using your software. https://www.youtube.com/watch?v=ubiiH3E55Y0
But I had a problem.
I often go out of town.
There is no internet and a router.
Can print directly?
From computer to Raspberry Pi?
Without the Internet and the router?

WesGilster, 2017-12-20:
I watched your video, but I'm afraid I don't understand the audio. :) Feel free to show off your printer in our forum. It seems like quite an interesting setup: https://forum.photonic3d.com/

I travel a bit myself in rural Iowa and I usually debug Photonic3D in the car without any network. Is there a feature that isn't working for you? If you are wondering how to use Photonic3D without a network, just type this in your browser: http://localhost:9091

At this point Photonic3D should work without a network connection, without printer firmware(recent change) and on any OS that supports a JVM.

olegmorgun, 2017-12-24:
Thank you!
Happened!
It seems that "Anti-aliasing" does not work.
When printing directly from a computer - The steps were not visible...

WesGilster, 2017-12-25:
Were you looking for the url? Installation steps? What kind of steps did you want?

olegmorgun, 2017-12-26:
[image]

olegmorgun, 2017-12-26:
Steps

WesGilster, 2017-12-26:
I see, I thought you wanted some kind of installation steps for offline printing. So you say printing from the Photonic3D/Raspberry Pi vs. Photonic3d/computer broke Anti-Alias?

The only difference between your two prints was the Raspberry Pi?

olegmorgun, 2017-12-27:
Yes

WesGilster, 2017-12-27:
Cool, I'm pretty sure I know what that was, if you were using the "Force Raspberry Pi" driver. I believe this is fixed in the development version of Photonic3D. The alpha encoding was broken until recently.

WesGilster, 2018-01-21:
Closing since offline printing was done long ago.

closed 2018-01-21 by WesGilster