area515/Photonic3D #348: Printer set up

tjfreckles, 2018-05-21:
Ok, I'm sorry if this isn't exactly at the level of most people on these forums but here goes :)

I'm making an SLA DLP printer with a Nema 17 stepper motor and a projector.

I've downloaded photonic onto my raspberry pi and can connect to he gui via google chrome, which is great.

Question 1: should my raspberry pi when I run it with a projector still show the desktop like before i ran the commands, or should it have pulled up a gui of some sort. That's what I originally thought was going to happen before but I just wanted to double check, because right now hooking it into the projector it's just going to a desktop

So assuming that everything is installed correctly I need to try and set up my stepper motor. I'm running an arduino uno  with Grbl connected to a stepper driver A4988 PLOU into a Nema 17 motor. So from the pi my connections are: PI > Arduino > driver > motor. 

Question 2: Do I need ramps to make the arduino run the stepper driver? 

so if I don't need any further hard ware my final question becomes about setting up the printer in the GUI of photonic. I selected:
 simulated display (not sure what that means)
Always full screen mode (if that's relevant idk)
GCode for the serial port
Auto detect projector (I also tried a few other settings for this one since it said it wasn't detecting it)

After the printer is set up, I constantly get the message saying my printer isn't started if I try to move the stepper motor. 

When trying to start the printer it says it can't determine the projector model. It's a Benq projector that I'm currently testing with. When I don't select a setting for the projector however it does work, however trying to move the does not work. 

Question 3: What do I need to do, or what information do you need to be able to help me. I'm sorry for being an idiot but I have no clue what I'm doing

jmkao, 2018-05-21:
We can help, but the forum is a better place for discussion outside of the software itself. https://forum.photonic3d.com

tjfreckles, 2018-05-21:
Ok jmkao, I just posted to forums now, but I apparently still need approval for my post to show up.