area515/Photonic3D #352: Is the project dead?

X3msnake, 2019-02-28:
We really need a Open Sorce alternative to Mango and NanoDLP

I see that there hasnt been any updates recently is the project dead?

kloknibor, 2019-03-06:
It hasn’t been really active lately in the main branches. There is some activity in forks.

Currently I didn’t have a 3D printer so my motivation slipped away. But I might get intrested in it again :-)

X3msnake, 2019-03-06:
what happened to your printer?

kloknibor, 2019-03-09:
I have been moving around, but now I have a permanent place with space for 3D printers again :-)! But it still might take a bir before I start woth this project again

kloknibor, 2019-03-09:
Also Wes was the main dev, I was more a helper ;-)! But it’s a good learning project for me!

X3msnake, 2019-03-09:
what happened to the main dev?

the community really needs a open source alternative to nanodlp and mango

i have high hopes this could be the start of it.

can you share what would help in getting you and the team moving further on
this project, what's missing? Parts? People? Know how? Drive?

kloknibor, 2019-03-13:
Mainly drive I suppose. 

One of the most active devs was @WesGilster but I assume he is busy with other stuff. If you are a company you can try to contact him to see if he can be motivated with a paid project/addition to Photonic if you have special wishes.

Also a while in the past @jmkao was involved with kudo3D and was quite active. He was also a great addition to the dev team. And in the early days there was another guy that started the project with Wes but I personally didn’t interact with him... I was more of a trainee than a real dev, although I know my way around the software, debian and I can do quite some work on the front-end. But since Wes was mainly doing the backend and my java being limited it also isn’t motivating for me. Lately I’m more focussing on automating my home with homeassistant and learning python to make add-ons for that. And also my work consumes a lot more of my time, so that’s where my time goes currently ;-) 

I must say it is quite common with opensource projects that the drive and with that the project dies if the dev community doesn’t grow for a project, there haven’t been a lot of PR’s from other people than the guys I mentioned.

X3msnake, 2019-03-13:
Not a company, and i don't care about comercial projects unless they are Open.
I belong to a fellow Hackerspace altLab.org, but I have been talking to some Masters in the industry that want to lauch a fully open source Firmware/Hardware/Software Solution for LCD printers. The moment is right for Open Source. 

I started the photonsters.org github and it atracted a team of talented devs there and now with a project to open up industrial abandonware and collect open source plans for open manufacturing, be it machines software firmware or simply methods and industry best practices, and bring them in a curated form to the Open Source Community www.openmachines.xyz

WesGilster, 2019-04-28:
Sorry it's been so long since I've posted. Let me know how I can help. Here are some development tips here:
https://wiki.photonic3d.com/doku.php?id=developers