area515/Photonic3D #353: KLD-LCD1260 Profile

portalro, 2020-08-04:
I would like to have a standard configuration to KLD-LCD1260 because it is very hard to find a tutorial for who it is starting with SLA configuration.

It is possible to update that?

blackrek, 2022-04-22:
@portalro, did you manage to set it up?

portalro, 2022-04-22:
No, since a posted this comment my printer is without to work. I am not finding the right configuration for work properly. Do you have this setup?

portalro, 2022-04-23:
Ok, if you get success please tell me, I never used my printer

blackrek, 2022-05-27:
So, as it turned out, I got a display with a broken cable, then I ordered and waited for a new display (whose protective glass turned out to be thicker than necessary, but this can be fixed by raising the limit switch and bowl), I also had to order a new mipi converter .

It came and I spent a lot of time trying to run it on raspberries.

Today was the first success, but with nanodlp.
The z-axis moves as it should (but you still need to install the patched marlin firmware), UV also turns on and off. After that, there were many experiments with the raspberry image and the config. In the end, I installed the nanodlp version on the desktop and was finally able to adjust the display timings so that it worked clearly and without flickering.
I still need to do some testing before the first print, and also get the invisible edges of the display properly adjusted, but so far everything seems to indicate that the printer will work.
As soon as I can print something intelligible, I will definitely describe the sequence of actions in more detail if it is interesting for nanodlp. However, my config may differ from the one you need, because. you most likely have a different display with a converter.
Here's what it looks like so far. A model with vertical stripes had to be made for testing.

[image]
[image]

portalro, 2022-05-28:
Wow, I am so happy for you! Congratulations.

Please, share your progress with me, I am so excited to make my
printer works.

Thank you to share.

Paulo

blackrek, 2022-05-29:
Received the first print. The first time the quality was good.
I don’t remember the exact sequence, because. there were a lot of robberies.
This repository is not the right place, but no one supports it anymore.

Here are the approximate steps:
1. Got a working config.txt. There are probably more unnecessary settings here, such as overclocking and screen rotation, but these work for me and I don’t touch them.
```
hdmi_pixel_freq_limit=300000000
hdmi_timings=1440 0 70 35 45 2560 0 12 2 2 0 0 0 37 0 151550000 0

hdmi_group=2
hdmi_mode=87

hdmi_drive=1
disable_overscan=1
max_framebuffer_width=1440
max_framebuffer_height=2560

config_hdmi_boost=4
dtoverlay=vc4-fkms-v3d
gpu_mem=256
gpu_freq=525
over_voltage=3
display_rotate=0x10000
hdmi_force_hotplug=1
hvs_priority = 0x32ff

framebuffer_ignore_alpha=1
framebuffer_depth=24
```

You will most likely have different settings for timings and screen frequencies.

2. I installed nanodlp on a ready-made raspberry os system using a script from the official site, but it is likely that the finished image from nanodlp will work.
When setting dtoverlay=vc4-fkms-v3d, I displayed only part of the desktop screen (with nanodlp autorun turned off), but nanodlp itself displayed images correctly. When installed, it will place itself in autorun. I added the service manually and for this I also had to disable the cursor
sed -i -- "s/#xserver-command=X/xserver-command=X -nocursor/" /etc/lightdm/lightdm.conf

3. For more correct operation and in order for the [[WaitForDoneMessage]] command to work in nanodlp, I flashed the controller with firmware from here
https://github.com/WheresWaldo/Marlin_KLD-LCD

In order not to poke around in the pins, the configuration suited me
#define MOTHERBOARD 35
The optical limit switch, motor control and UV on/off work correctly with it.

I am attaching the files of the machine and settings for burning nanodlp. This is the first version, but it suits my resin, maybe I will correct something in the future.

[nanodlp-kld1260.zip](https://github.com/area515/Photonic3D/files/8793339/nanodlp-kld1260.zip)

problems:
sometimes after starting nanodlp does not display images on the screen, restarting helps.

My first probe (bottom):
[image]

Enjoy!
If you need any details, please contact me - I will help in any way I can.