area515/Photonic3D #354: Linux Install destroys Config. NO WARNING

TecDroiD, 2021-03-25:
i installed your product on my system not knowing that it completely destroys configuration.
There was no warning on your page and i am unable to work anymore.
What the **** do yo do in your scripts and how can i uninstall your shit?

thetrebor, 2021-03-25:
Not a maintainer: Presumably you just ran start so it does this:
https://github.com/area515/Photonic3D/blob/master/host/bin/start.sh

looks like it installs java JRE and updates init.d and such to make it easy to put on a RPi.  Guessing you ran this on your workstation in not a VM or an RPI.  Maybe you could reverse the changes in that file and hope for the best.  Good luck!

thetrebor, 2021-03-25:
> problem is, it always runs a root desktop when i restart the machine.
> cannot get back to my normal workstation desktop.
> What did they do to do that?
>
> I didn't install a Raspberry Pi version so my believe was that i get a
> background process which i can use in a browser.

Yeah this is control software for a printer not a remote access system for
printers.  Once it’s installed on a printer and configured properly then
you’d use a browser from your desktop to access it.
Your desktop thinks it is a printer now.  You’ll probably need to boot to
single user mode and undo things.  Sorry man.  Good luck.

TecDroiD, 2021-03-25:
which things?

thetrebor, 2021-03-25:
Like I said, not a maintainer so I’m not entirely sure.  But probably you
need to get your init.d stuff restored.
Probably this just dropped a couple files in startup that take over.  Look
through that startup script and see what it copied into your initialization
and comment them out.
Hopefully it didn’t overwrite anything.

sobryan, 2021-03-25:
According to the start script:
https://github.com/area515/Photonic3D/blob/master/host/bin/start.sh : line 172
This is run:
update-rc.d cwhservice defaults

To remove this, I think you want to execute this as root to remove the service and restart
sudo update-rc.d -f cwhservice remove

TecDroiD, 2021-03-25:
i basically did this but by hand. Seems, systemd runs init scripts to create virtual service units.. this way, it already started x..
Why does this call startx? all the init script should do is running the jar. 
To my mind it's an absolute nogo to run anything as root. Especcially on anything else than a raspberry pi.

sobryan, 2021-03-25:
Good deal - what worked for you (in case someone else needs this)? 
I think would probably work as well
chmod -x /etc/init.d/cwhservice

On quick glance, I'm guessing startx is called to create a display that the java application can use.

WesGilster, 2021-03-25:
It is needed for Java but that's not why Photonic required it. It is required if you want to use the embedded browser to control the printer. If you'd like to support multiple screens(LCD screens) with the Raspberry Pi, your options are somewhat limited. The route we took was:
1. X Windows (which doesn't support multiple monitors on the Raspberry Pi)
2. Low Level DispmanX drivers for rendering (requires Root)

sobryan, 2021-03-25:
Added a note to the linux section as suggested - hope this helps! 
d5b03d56c2838120163db7e33aaeb5095eedb660
c8ee714da09b3ac207c61887ee44496178c70f7f