area515/Photonic3D #357: Creality LD001

errolt, 2022-03-08:
Hi whomever is still active here,

I was looking through a software update for the Creality LD001 and found Photonic3D assets in there. Did Creality straight out rip this project off? Or did they buy the code and close source it?

Also, anyone know what happened to Creation Workshop?

Just wondering...

Thanks,
Errol

kloknibor, 2022-03-08:
The project is opensource so they probably used this code for their product. If they did use this as an source it would not be allowed to close the source.

The license provided with this project is the following:

https://github.com/area515/Photonic3D/blob/master/LICENSE.md

What happened to CW I'm not sure. The last I know of is that this code did get bought and the people trying to make it profitable kinda failed to keep interest.

errolt, 2022-03-08:
Unfortunately they did close source it. 
And they locked it down. 
The SD card is locked to the Pi. Can't replace the Pi if it dies, and can't use someone else's SD image if your SD card dies.

Anyway, just FYI.

closed 2022-03-08 by errolt

WesGilster, 2022-03-09:
Would you like to tell us a bit more about which references you've found to photonic3D? I'd be happy to start a conversation to determine if their implementation is fair use.

errolt, 2022-03-09:
Hi @WesGilster 
My first reference that led me to this repo was an image in the firmware update under /resourcesnew/cwh/img, which was this image:
https://github.com/area515/Photonic3D/blob/master/host/resourcesnew/cwh/img/photonic3d-icon-sign.png

There is also a files under /lib/ called photonic3d.jar and photonic3d-tests.jar

If you want to take a look then you can download the firmware for LD-001 from https://www.creality.com/download/
The file is called LD001-0.4.7-release-0.smu
It is a 7z archive, and I found the password for the 7z file in the readme here: https://www.dropbox.com/sh/9eqqny9e2jqq362/AAAh3YyaepN0y5vnrB78VVHja/firmware%200.4.5.%20-%20patch%201?dl=0&preview=readme.txt&subfolder_nav_tracking=1

I doubt you will get anywhere with Creality though. 
Not only did they close source it, they locked the PI3 SD card to the PI it shipped with. So you can't replace your Pi3 if it dies, and you cant use an SD card from another printer if your SD card dies.

kloknibor, 2022-03-09:
@WesGilster I just checked the folder as @errolt suggested. It 100% uses Photonic3D resources:

[image]

[image]

The manufactorer seems to  be from china though, the location on their website is this one:

1202, Block 3, JinChengYuan, Tongsheng Community,Dalang, Longhua District, Longhua District, Shenzhen,China, 518109

They do sell mainly in EU,AU and UK. 

If there is anything I can help with from the EU or UK front since I'm located in these areas let me know. But since the manufacturer is chinese and they are not that strict with copyright not sure much can be done...