area515/Photonic3D #359: projector

elliot-mcallister, 2022-10-19:
Hi, I am writing- and I don't know if this project is dead or not so no worries if it is not supported. - I am building a DLP printer. I had it working on nanoDLP - but need to switch programs to be able to review what is happening on the printer. I came upon photonic, and installed. I have everything running except it would connect or does not understand the projector model. Is there any possibility you might be able to help?

mUVe printer
RasPi4
Optoma UHD60 (4k)
Arduino/Mega 2560
Marlin V2.0 (have 1.0 too if that is needed)
DL from https://github.com/area515/Photonic3D

Thanks,
Elliot

WesGilster, 2022-10-20:
You will need to build your own printer profile if you understand the commands your printer requires for each function. Place those gcode commands in those functions and away you go. It's not really that necessary to pick your project model. That only when you some advanced functions like controlling your printer to turn on/off through it's serial port.

elliot-mcallister, 2022-10-21:
Hi Wes,
Thanks! I understand the printer commands and gcode. I have a mUVe printer
that is heavily modified and was running on nanoDLP.

What I seem to be missing is the proper config file for the projector  +
raspios interface. I had this running on a ras pi3 with (noobs or
jessielite), Optima uhd60 (4k), and RAMPS 1.4, MARLIN 1.0.

I'm trying to upgrade it to Raspi4, bullseye-lite, Marlin 2.0 or 1.0 (I
have both working on NanoDLP).

It seems to be the config file,  because I can get the printer online and
interact with the GUI. It recognized the ramps board but I get an error on
recognizing the projector.

I'm happy to share the config files etc

Any help is greatly appreciated.

Thanks!

E

elliot-mcallister, 2022-10-21:
> I wouldn't choose Automatic Projector detection. I would either use the
> RaspberryPi screen or even the Xwindows driver if it's detected. If you
> want to use the RaspberryPi drivers you need to make sure your
> configuration has them enabled:
>
> displayDevice.org.area515.resinprinter.display.dispmanx.RaspberryPiMainLCDScreen=true
>
> displayDevice.org.area515.resinprinter.display.dispmanx.RaspberryPiForceTVScreen=true
>
> If you still have trouble detecting a screen, you'll need to start using my
> repo instead of area515 since it's quite a ways behind.

Thanks Wes! I'll try your repo.

Best regards,

Elliot

elliot-mcallister, 2022-10-21:
Hi Wes, Thank you for the help. If I switch repos I noticed you had a -Dev
and a regular flavor. Is there any difference - would one be better for a
DLP vs LCD?

Thanks again.

Best regards,

Elliot
_________________________
phone: 540.759.2447
________________________

CONFIDENTIALITY NOTICE:

This electronic mail transmission is confidential, may be privileged and
should be read or retained only by the intended recipient. If you have
received this transmission in error, please immediately notify the sender
and delete it from your system.cell: 540-759-2447

elliot-mcallister, 2022-10-24:
Hi Wes,
I noted a -Dev and the other repo on your list is one more up to date?

I tried the wget command but my connection was refused. I will try again
later today.

BTW, is it necessary to use wheezy still or is the new RasPiOS ok? I had
installed BullsEye with the java modules to ge this far on the 515 repo.

Many thanks,

Best regards,

Elliot
_________________________
phone: 540.759.2447
________________________

CONFIDENTIALITY NOTICE:

This electronic mail transmission is confidential, may be privileged and
should be read or retained only by the intended recipient. If you have
received this transmission in error, please immediately notify the sender
and delete it from your system.cell: 540-759-2447

elliot-mcallister, 2022-10-25:
Hi Wes,
I attempted to load this to my pi, and have a number of errors. I am happy
to send the text of the messages but I may be missing something from the
git repo: I am not a coder by any means and have little experience in
RasPi. Any questions here that seem idiotic are because I have so little
experience.

Thanks!

1. what java version are you using or does the program require - it looks
like it is searching for a graphics packet for the config file but is not
in the directed location?
2. seems that some items have been moved I get a 302 error - looks like
they have been resolved by forwarding messages
3. missing startx and hostx commands, not sure if these are important
4., jdk-8u101-linux-arm32-vfp-hflt.tar.gz': No such file or directory - Is
the repo for 32bit only?
5. tar (child): jdk-8u101-linux-arm32-vfp-hflt.tar.gz: Cannot open: No such
file or directory
tar (child): Error is not recoverable: exiting now
tar: Child returned status 2
tar: Error is not recoverable: exiting now
A new version of Java is available, please update this script with the
proper download URLS from:
http://www.oracle.com/technetwork/java/javase/downloads/index.html

Best regards,

Elliot
_________________________
phone: 540.759.2447
________________________

CONFIDENTIALITY NOTICE:

This electronic mail transmission is confidential, may be privileged and
should be read or retained only by the intended recipient. If you have
received this transmission in error, please immediately notify the sender
and delete it from your system.cell: 540-759-2447