area515/Photonic3D #360: Request for Packaged Distribution of Photonic3D (Nixpkgs, AppImage, or Flatpak)

mateowoetam, 2025-03-27:
Description: I would like to request that Photonic3D be distributed as a packaged application rather than relying solely on a shell script for installation. Packaging the software would enhance its accessibility, usability, and compatibility across various systems. Ideally, I suggest the following formats:

- Nixpkgs: This would integrate Photonic3D into the Nix ecosystem, allowing users to install it declaratively and reproducibly.
- AppImage: A portable option that bundles all dependencies, making it easy to run on most Linux distributions without installation.
- Flatpak: A sandboxed format that ensures compatibility and security across Linux distributions.

Benefits:

- Simplifies installation for users, especially those unfamiliar with shell scripts.
- Allows users of Immutable Linux distros to have this software available to them.
- Increases adoption by making the software more accessible to a broader audience.
- Reduces dependency issues and ensures consistent performance across different environments.
- Aligns with modern packaging standards, enhancing the project's reputation and usability.

WesGilster, 2025-03-28:
I don't think anyone would be against it. Do you have experience with getting projects setup with these package managers? And if so, would to be interested in helping move that effort forward?

mateowoetam, 2025-03-28:
Not really experienced, but I could post this in [nixpkgs](https://github.com/NixOS/nixpkgs) to see of anyone is willing to do it.